ms-iot/samples #228: DHT 22 displaying wrong temp

zach78954, 2016-03-05:
After exceeding 25.5 C the sensor then reports 1.0 C. I have tested with two separate DHT22 sensors and I'm currently on build 14262. Has any one else had this problem?

bfjelds, 2016-03-07:
Does this refer to a particular sample?  Or have you written your own code to experience this?

zach78954, 2016-03-08:
This is using the GPIOOneWire example, with no modifications.

jordanrh1, 2016-03-08:
Hi Zach, do you have a code sample where this is working correctly, like on an Arduino?

zach78954, 2016-03-08:
No, unfortunately I have no other test setups.

jordanrh1, 2016-03-10:
I'm sorry that you running into issues with the DHT22. The sample was tested only with a DHT11. I've ordered a DHT22 and will update the sample when I can verify the fix with a DHT22.

zach78954, 2016-03-10:
Ok, please let me know the results of the tests. Thank you for your help. 

- Zach Stinnett

jasjitchopra, 2016-04-02:
Same issue here with DHT22 - I do not have DHT11 so cannot test the difference in the output between the two. Temp showing as 8-10 C whereas it should be between 30-33 C

Even the humidity levels show up as 10-20% whereas actual values are between 40-55%.

mr-mgut, 2016-04-10:
Hey guys, I had the same issue with DHT22 module (connected to RaspberryPi 2 Model B) - this is the bad news. The good news is that I managed to identify and resolve the bug that was causing this.
After reading a DHT22 datasheet (see https://www.adafruit.com/datasheets/DHT22.pdf) and scanning the code, it became quite clear to me that the bug is related to the invalid alignment of the sensor data. To be more explicit, the bug is located in GpioOneWire::Dht11Reading::Humidity() and GpioOneWire::Dht11Reading::Temperature() functions. 
In order for "((value >> 32) & 0xff) + ((value >> 24) & 0xff) / 10.0" expression to work we need to shift the bits of integral RH data by 5 bits to the left. So the correct expression will look like "(((value >> 32) & 0xff) << 5) + ((value >> 24) & 0xff) / 10.0". Moreover, we actually don't need to do this additional bit shifting. Instead, the expression can be written as "(((value >> 24) & 0xff00) + ((value >> 24) & 0x00ff)) * 0.1". As you can see, I also replaced the division with multiplication since it requires fewer clock cycles. You may also be wondering why are we multiplying the obtained value by 0.1. Well, this is strictly related to the humidity and temperature resolutions of DHT22 module which are 0.1%RH and 0.1Celsius respectively. Hope this helps :-)

IoTGirl, 2016-06-06:
Closing based on Fix by Jordan

closed 2016-06-06 by IoTGirl