ms-iot/samples #233: IoTCoreDefaultApp - DeviceType

grahamehorner, 2016-03-18:
at present the IoTCoreDefaultApp has hardcoded DeviceType enum for the hardware supported platforms, consider externalising this these values so as/when new hardware platforms become available a simple config file update is needed to:-

1. detected what hardware platform the application is running on
2. locate and load the appropriate hardware platform logo/display name
3. load the correct values for the GPIO pins using in tutorials etc.

bfjelds, 2016-03-21:
Thanks for the comment.  This is a great suggestion and we will add it to our backlog of items!

grahamehorner, 2016-05-18:
This could be as simple as providing a resx for the board named something like RaspberryPi.resx, RaspberryPi3.resx, MinoboardMax.resx etc. which can be dynamically loaded.

IoTGirl, 2016-06-03:
Feedback taken by IoT Team member - closed

closed 2016-06-03 by IoTGirl