ms-iot/samples #234: VideoCaptureSample not working with Pi 3

mikestu, 2016-03-19:
I bought a Lifecam HD-3000, since it looked like that web cam was supported by the Pi, but the VideoCaptureSample does not appear to work.  When I click to do the live video, all I get is a green color in the video area.  When I click to play it back, it still just shows green.  Am I doing anything wrong, or are there just things missing, to fully support this working with Raspberry Pi 3?  This is my first IOT purchase ever...maybe I should have just gone with Pi2.

MagicBunny, 2016-03-21:
Hi, Mike, sorry that you are having the issue. This sample was made for Pi2 and hasn't been tested on Pi3 yet. I will try it out on Pi3 and get back to you. Thanks for the feedback.

IoTGirl, 2016-06-06:
Confirmed with Maggie - Closing

closed 2016-06-06 by IoTGirl

mikestu, 2016-06-07:
I don't see that anything was changed for that sample, but this is now closed.  What does "Confirmed" mean...were you able to get it to work properly on your end?  Thanks!

IoTGirl, 2016-06-07:
The sample csProj states "Update SDK version to 10586 ".  This image does not support the Pi3 so this sample is not yet supported for Pi3.  Basically, if you are targeting Pi3, you are using a pre-release image and all samples are not expected to keep up with the myriad of changes in pre-release images.  Note that once a stable image is available, all samples will be run through a test pass for support on the release image.