ms-iot/samples #236: IoTCoreDefaultApp - Contribute Dutch (NL-nl)

Vinz68, 2016-03-29:
I have added the Dutch resource strings (and tested it on a PI2 + PI3).
I would like to contribute (and maintain) this to the IoTCoreDefaultApp, but I need some help how to do this since I am not allowed to upload a file.

Vinz68, 2016-03-29:
Ok, I just did an attempt to contibute the dutch resource file (and modified project file) after a quick online guide for Github. 
Looks OK so far. Hopefully it gets approved and merged into the master eventually.
Not sure if I need to assign a reviewer or team; but we will see...

bfjelds, 2016-03-30:
Thanks for the submission!!  That is awesome!  

I have started working with the MSFT localization team to understand what sort of review is required for this.

Vinz68, 2016-04-01:
I see that my contribution of the NL-nl resource file has been merged and the dutch language is now available.
 
Thanks !.

This issue can be closed then since it is solved.

closed 2016-04-01 by bfjelds