ms-iot/samples #262: IoT BACnetAdapter DSB sample target option

VinceT25, 2016-04-27:
Is it possible to add **ARM** to the build/deploy target list for the BACnet DSB sample? I would like to deploy it to an RPI3 device. Please advise.

Thanks,
VinceT25

zhuridartem, 2016-04-27:
This sample relies on the Open source BACnet Protocol Stack (bacnet-stack-0.8.2 from http://bacnet.sourceforge.net/), which had problems building for ARM and that is why we disabled that configuration.

I just tried with VS 2015 Update 2 to re-enable ARM configuration, and it built with warnings.
You can enable the ARM build configuration in VS by selecting Build -> Configuration Manager -> Active solution platform: <New...>, Type or select the new platform: ARM, Copy settings from x64.

This should build the project, but be warned that we have not done any validation to this sample for RPI3.

VinceT25, 2016-04-28:
Thanks for the tip Zhuri. I'll keep those caveats in mind as I proceed cautiously.
VinceT25

closed 2016-04-28 by VinceT25