ms-iot/samples #264: App2App WebServer blinky webservice not working anymore

Patafoin, 2016-04-28:
Hi,
I'm currently trying to use the blinky webservice to test communication between front GUI and back http server.
When I launch the C# sample, it crashes at the following line : 
builder.SetTrigger((IBackgroundTrigger)trigger);
With the following exception : Additional information: Specified cast is not valid.
I'm running it on a raspberry 3, with Windows 10 IoT version 10.0.14322.1000
My Visual Studio version is 2015 Enterprise edition update 2 (4.6.01038)
I tried it with Microsoft.NETCore.UniversalWindowsPlatform version 5.0.0 and tried updating it to 5.1.0, same problem.
Does anyone have the same problem ?

bfjelds, 2016-04-28:
Can you include more of your code?  For example, what is "trigger" ... how is it created?

Patafoin, 2016-04-28:
my bad, I think I have made a mistake modifying sample code thinking it was original code...
I just unzipped original zip and tried again, it works.
sorry for the useless post...

closed 2016-04-28 by Patafoin

bfjelds, 2016-04-28:
No problem!  We love any and all questions ... we are actively monitoring the samples Issues for any problems that people run into when trying to apply the samples to their real-world scenarios.