ms-iot/samples #275: Temp calculation - much too high

torbenvaering, 2016-05-18:
Thanks for a great example!

The app reports
819.2RH
537.6C
any idea how to get some more realistic numbers?
OS is 10.0.14328.1000

Thanks in advance.

IoTGirl, 2016-06-03:
Hi Torben,
What does the sensor manufacturer say about the return value? Do you need to parse it in some way? 
Sincerely,
IoTGirl

TGoodhew, 2016-06-03:
Hi Torben - Justine mentioned your issue to me and I vaguely remember having a similar issue for another project I worked on.

It was related to the actual conversion of the value read from the sensor to a real value. Can you point me to the exact piece of code that is causing you the problem? 

Justine said that it was the old Galileo sample for SmartFan - I still have the working code there that I forked off for my own use so if that is it then I can share.

Please let me know.

TonyG
Program Manager, Windows IoT Core.

torbenvaering, 2016-06-05:
First of all, thanks for helping me!
I have no clue where it goes wrong. Sorry. I am a total newbie when it comes to win iot.

I just decided to start of with an easy example, and purchased the parts (mostly from eBay).

I have included a picture of my DHT11. I thought it was one vendors item name, and not a generic name.

I have tried different environment, and even exhaling on it, just to see if the values at least _changed_ - and they did.
It is a temp and humidity example: github.com/ms-iot/samples/tree/develop/GpioOneWire
Does that have something to do withSmartFan and Galileo?

[image]

TGoodhew, 2016-06-05:
Ahh – This doesn’t have anything to do with SmartFan & Galileo.

It’s not unheard of for components of EBay to be broken but Aosong is the original vendor of the DHT11 sensor.

This sensor uses the Dallas Semiconductor OneWire protocol – It’s unfortunate but we don’t have native support for that in IoT Core – How the sample is working is to basically trying to read the pulse width and then calculate a 1 or 0 from the time between two falling edges.

This makes it very sensitive to the performance of the RPi.

What I would do is reflash the SD Card with the current RTM version of IoT Core (not the insider version) and then go into the web admin tool and turn on the DMAP driver (Details here https://developer.microsoft.com/en-us/windows/iot/win10/lightningsetup) – This driver provides a much faster GPIO access so there should be less overhead to the Pi polling to find the falling edge.

If that doesn’t help then you’ll need to put an oscilloscope on the sensor output to measure the actual signal being sent back from the device and confirm that it is returning valid data. If it is then you’d need to debug into GpioOneWire::DhtSensor::Sample() to see what the code thinks is being returned.

If you don’t have a scope and the lightning driver doesn’t help then I’d suggest changing the sensor to an I2C or SPI one. Personally I like this guy:

https://www.adafruit.com/products/2652

But there are many options.

If you really want to try and get the DHT11 to work then you could use our implementation of Arduino Wiring and try getting the Arduino code located here working - https://learn.adafruit.com/dht

If none of this works for you then let me know and I’ll see if I can dig up a DHT11 and put it on my scope to see whether there is just an issue with our sample.

TonyG

[dht11]<https://cloud.githubusercontent.com/assets/19437803/15807415/12b907ca-2b5e-11e6-9285-24e69b4fae6e.png>

EdiWang, 2016-06-06:
I use code from https://www.hackster.io/porrey/dht11-dht22-temperature-sensor-077790?ref=part&ref_id=9417&offset=15 seems to be accurate on my RPi3

IoTGirl, 2016-06-06:
There is a previous temp issue as well. See https://github.com/ms-iot/samples/issues/228

TGoodhew, 2016-06-06:
I also just heard that there is a performance regression in the .NET Native Toolset – I don’t know much more about that but you should try these two things first:

1.       Turn on the Lightning driver

2.       Turn  “Compile with .NET Native tool chain” on the Build tab of the project properties

If they don’t help then try the other stuff I mention.

TonyG

https://www.adafruit.com/products/2652

torbenvaering, 2016-06-06:
Hi TGoodhew (and other who helped),

I have installed the Lightning driver, but that only results in no data being read. So I'm ready to try the Native Tool Chain, but I can't find the tabs and menu you describe. I have Googled it, and can recognize what it looks like, but I can't get to it. This is what I see when I right click on the project name in the solutionexplorer:
[image]

[image]

TGoodhew, 2016-06-06:
Ahh – Sorry I thought you were using your own C# version of the sample - .NET Native doesn’t apply to the C++ case.

When you change to lightning you’ll need to change how you get the default controller. You do it this way:

void MainPage::InitGPIO()
{
    if (LightningProvider::IsLightningEnabled)
    {
        // Set Lightning as the default provider
        LowLevelDevicesController::DefaultProvider = LightningProvider::GetAggregateProvider();
    }

    auto gpio = GpioController::GetDefault();

    if (gpio == nullptr)
    {
        pin_ = nullptr;
        GpioStatus->Text = "There is no GPIO controller on this device.";
        return;
    }

    pin_ = gpio->OpenPin(LED_PIN);
    pin_->Write(pinValue_);
    pin_->SetDriveMode(GpioPinDriveMode::Output);

    GpioStatus->Text = "GPIO pin initialized correctly.";
}

From here: https://github.com/ms-iot/BusProviders/tree/develop/Microsoft.IoT.Lightning.Providers

Basically if you don’t get the lightning provider then you’ll get nothing.

TonyG

[udklip0]<https://cloud.githubusercontent.com/assets/19437803/15836911/4d38e300-2c37-11e6-9442-a5a46213344d.PNG>

[udklip]<https://cloud.githubusercontent.com/assets/19437803/15836919/535cc1c0-2c37-11e6-9ab7-655e813bed1d.PNG>

torbenvaering, 2016-06-08:
Thanks Tony! Where do I put that piece of code?

TGoodhew, 2016-06-08:
It isn’t a cut and paste – What I was showing with that piece of code is how you get the lightning controller for GPIO.

If you look at the GpioOneWire::MainPage::Page_Loaded() you’ll see where it creates the controller object:

GpioController^ controller = GpioController::GetDefault();
if (!controller) {
this->statusText->Text = L"GPIO is not available on this system";
return;
}

You want to replace that with the patter of:

1.       Check if the lightning driver is enabled

2.       If so then set the low level default controller to be the lightning controller

3.       Get the GPIO controller

4.       Party on controller like normal

Basically this code ensures that you’ll get the lightning controller if it is present and the normal one if not.

TonyG

torbenvaering, 2016-06-08:
Thanks again Tony. I really appreciate you taking the time.

I don't think I'm going to do this project anyway. It is just too complicated for me. What I don't understand is why MS present this as something that is ready to go. I don't understand how the original example was able to produce a valid measuring.
But really. Thanks for helping!

TGoodhew, 2016-06-08:
This is a case where there are tight tolerances for the digital signal transitions and, unfortunately things can fall far enough apart that the same doesn’t work.

We have addressed this in the next version of Windows 10 IoT Core by enabling interrupts on digital pin transition so it makes the timing aspect much simpler and more reliable.

Personally I use I2S or SPI based sensors for temperature – If I’ve already included an ADC in my project then I’d use something simple like a TMP36 analog temperature sensor.

TonyG

IoTGirl, 2016-06-20:
Hi torbenvaering,
I am sorry to hear that you have abandoned this project but I do hope you start another!
As there have been no more questions from you, I am closing this issue.
Sincerely,
IoTGirl

closed 2016-06-20 by IoTGirl