ms-iot/samples #278: samples/BluetoothGATT/CS work not allright with my device

jasonbu, 2016-05-26:
I'm trying to edit the code to fit my device which using nRF52832, advertisement and connect is OK, but service discovery got some trouble, 

callback is arrived but the service is null ... no idea why.

what i edit is only the service uuid

jasonbu, 2016-05-27:
I found it only happen when i debug the x86 on my x64 system

IoTGirl, 2016-06-07:
Hi Jason,   There is really not enough detail here to move forward.  Can you try and provide more information?  At Microsoft we collect all the information needed to see the issue here and I have a simple template that might help you provide the detail we need. 

**ENVIRONMENT:**
 
What Computers, OS and devices are involved with this repro? How are they connected?
 
**REPRO STEPS:**
 
What steps would someone else need to follow to see the issue?
 
**RESULTS:**
 
What did you Expect to happen?
 
What did Actually happen?
 
**NOTES:**
 
Any other details/workarounds potentially related info you can provide?

Sincerest thanks,
IoTGirl

jasonbu, 2016-06-08:
tks for reply
ENVIRONMENT:

Win10x64 10586，ble work with usb bcm20702, 

REPRO STEPS:

clone the repo, work with samples/BluetoothGATT/CS,debug with local machine,tried x86 and x64 mode

RESULTS:

What did you Expect to happen?
I think x86 should work all right with x64 machine

What did Actually happen?
x64 work perfectly and x86 got null service pointer

NOTES:

as a x86 environment, it's unable to watch the variable in vs2015, but it seems to get a null pointer when service discovery finish, and in the gatt discovery finish callback. x64 work perfectly

IoTGirl, 2016-06-08:
These samples are targeted for IoT Core of which there is no x64 version.  Can you tell me which board you are using? MBM, RPi or DragonBoard 410c?

jasonbu, 2016-06-17:
just pc... what amazing uwp...

IoTGirl, 2016-10-31:
Sorry Zeeshan, I have no BT hardware.

IoTGirl, 2016-11-02:
Hi JasonBu,

I have some more information for you:

- There are some known issues with Bluetoooth GATT on IoTCore ( See [https://social.msdn.microsoft.com/Forums/en-US/3c6bb41c-6f1d-4e11-ba3a-fac208721bc7/new-flight-14955-released?forum=WindowsIoT](https://social.msdn.microsoft.com/Forums/en-US/3c6bb41c-6f1d-4e11-ba3a-fac208721bc7/new-flight-14955-released?forum=WindowsIoT )) 

- These samples are focused and tested on platforms with IoTCore (MBM, RPi & DB410c) and that is only shipped with X86 support though I believe x64 can be created by an OEM

Can you confirm with the latest version of Windows 10 (14393) and hardware you are using?

Sincerely,
IoTGirl

jasonbu, 2016-11-03:
um... I'm just using desktop computer with x64 14393 Enterprise now,  when i found the trouble I'm using 10586 x64 Enterprise desktop... maybe it's not tested with my platform

IoTGirl, 2017-01-19:
Hi Jason,

I am sorry I don't have a better answer for you but please take a look at the RFCOMM sample at [https://github.com/Microsoft/Windows-universal-samples](https://github.com/Microsoft/Windows-universal-samples) as that should be directly related to the desktop.

You are correct that UWP should work across the OS platforms but not all flavors of Windows 10 have the same API surface and the same capabilities available.  This repository is supported by the IoT Core Team but the repository above is supported by the folks you want to leverage.  Please try the RFCOMM sample and give feedback as to your experience on that sample share.

Sincerely,
IoTGirl

closed 2017-01-19 by IoTGirl