ms-iot/samples #284: Why do we have two SSD1306 samples?

stefangordon, 2016-06-05:
I'm confused why we have this separate sample for the SSD1306 display via SPI accepted? https://github.com/ms-iot/samples/pull/113

I think this pull request was accepted in error - not only is it confusing to users because we already have the "SPIDisplay" sample (which is already an SSD1306 SPI driver), but the folder structure of the accepted PR is completely out of whack with the rest of the repository.

EdiWang, 2016-06-06:
I agree. I am also looking for the I2C version instead of SPI version of SSD1306, but I can not find it anywhere, seems nobody has written code for this on Windows 10.

stefangordon, 2016-06-06:
Yes I actually ported it to an i2c version, which is why I was investigating current state.  I'll get it put up somewhere later today and update here with a link @EdiWang

stefangordon, 2016-06-06:
@EdiWang I pushed my i2c sample here
https://github.com/stefangordon/IoTCore-SSD1306-Driver/tree/master/ssd1306/Display

It is basically the SPI sample but ported to I2C.  I've enhanced it with the ability to define images (of arbitrary size) in addition to fonts, and added a bit of inline documentation to display.cs.

I also created a little web app you can use to design images for the screen
http://dotmatrixtool.com/

I'll get this all cleaned up and figure out how to merge in later, but feel free to use it for now.

EdiWang, 2016-06-06:
WOW! Thanks very much @stefangordon 

That's exactly what I am looking for!

sely2k, 2016-06-07:
hi @stefangordon the two version is because the "original version" of the sample is not well fit for the oled display 1306 with 64 line

stefangordon, 2016-06-07:
May be best to improve the original sample as it is confusing this way.  If you are able feel free to assign this issue to me and I'll try to work on it in a few weeks.

IoTGirl, 2016-06-08:
You are "Assigned". Thanks Stefan!

stefangordon, 2016-12-05:
My fork is still there for people who want an i2c version, but I have not had a chance to integrate into the repo yet.  Given it isn't actually a bug we can probably close the issue if appropriate.

sukeshak, 2016-12-09:
Thank you @stefangordon 

I found this implementation very interesting since it uses System.Drawing namespace... do you guys think this method has any downside to it?
https://github.com/jakkaj/xIOT/tree/master/Samples/OLED_SSD1306