ms-iot/samples #286: Bluetooth device list shows duplicates in IoTCoreDefaultApp

mjmeans, 2016-06-06:
1. Connect using Windows IoT Remote Client (Preview) to IoTCoreDefaultApp in system using CSR 4.0 Bluetooth adapter.
2. Click on Settings and then on Bluetooth. Turn Bluetooth on. Observe list of devices able to pair.
3. Click the <- (back) button. Click on Settings again. Observe that the same devices are listed, but now they are listed twice.
4. Repeat #3 multiple times. Each time it is repeated more duplicates of the same devices are shown.

IoTGirl, 2016-06-06:
Hi mjmeans,
Can you please confirm the devices involved?
Sincere thanks,
IoTGirl

mjmeans, 2016-06-06:
Raspberry Pi 3 with CSR 4.0. The Bluetooth devices in question are: "[TV] Samsung LED46" which is actually a Samsung TV; and the other device is a laptop (Dell Inspiron 7537) which has built-in Bluetooth. I am available for a remote screen connection if you need to actually see it happen.

IoTGirl, 2016-06-06:
Hi mjmeans,   I have asked a member of the IoTCore Team to take a look. Sincerely, IoTGirl

IoTGirl, 2016-06-20:
Fixed

closed 2016-06-20 by IoTGirl