ms-iot/samples #306: IoTBrowser

rudman11, 2016-07-23:
Is there a way to set the proxy settings for the IoTBrowser running on RasPi 3 using Windows IoT Core?

IoTGirl, 2016-07-25:
I suspect you will need to use a UWP version of WinHTTP.SetProxy.  Here is a previous version of the documentation [https://msdn.microsoft.com/en-us/library/windows/desktop/aa384069(v=vs.85).aspx](https://msdn.microsoft.com/en-us/library/windows/desktop/aa384069(v=vs.85).aspx)

IoTGirl, 2016-12-05:
Hi Rudman,
The team has accepted this request as a work item and have added it to their backlog. 

> We are working on support for configuring the proxy on IoT via netsh.exe command.

 (9113049)  

As this team works on the agile model, I can't give you a timeline for this feature to be implemented.

Sincerely,
IoTGirl

closed 2016-12-05 by IoTGirl

IoTGirl, 2017-01-12:
NOTE: **Add WinHTTP netsh helper for proxy configuration in IoTCore** images was accepted to address this request.