ms-iot/samples #317: DigitalSignageUAP

dunnleaddress, 2016-08-12:
on line 69 of MainPage.xaml.cs:
"if (time.Hour == 0 && time.Minute == 0) // we want the content reload happens only at 0:00 every day, exact mid-night"
I made changes to config.xml however the contents did not get reloaded at mid-night. Can you please advise me asap? Thanks so much!

nicogbg, 2017-02-17:
since DispatcherTimer aren't guarantee to fire immediately after interval expires and also because
it started some when seconds within a minute there is no guarantee that time will exactly be 0:00 when reloadContentTimer_Tick will be called. You should allow a time window around midnight in the if statement (and debounce).

zhuridartem, 2017-03-31:
This issue should be fixed in latest version of the code.

closed 2017-03-31 by zhuridartem