ms-iot/samples #325: documentation issue

cyberh0me, 2016-08-25:
reference https://developer.microsoft.com/en-us/windows/iot/samples/zwaveadapter

it seems the documentation does not reflect the code changes, for example
 
"Once the solution has been opened in Visual Studio, Navigate to the Solution explorer and right click the ZWaveBackgroundService project. Select “Set as Startup Project”. "

but the current solution does not include a ZWaveBackgroundService ... and also the screenshot shows "other" projects included in the solution

please update the documentation, thank you

steven045, 2016-08-26:
I am having a similar issue. When I open the ZwaveAdapter solution from Visual Studio 2015, I don't see ZwaveBackgroundService. I See something completely different.  The tutorial mentions to set ZwaveBackgroundService as startup project. Has something changed from the tutorial?

IoTGirl, 2016-09-01:
This should be addressed with @jhutchings1 merge.

closed 2016-09-01 by IoTGirl