ms-iot/samples #342: Blocky not working

jwillmer, 2016-09-30:
## Setup

- I have a Raspberry 3 with sense hat
- I deployed the latest Windows IOT on it. 
- I used the IOT Dashboard to deploy blocky.

## Problems

- With the stable Windows IOT image I can't reach the web interface of blocky - config interface is available
- With the preview image I can reach the web interface of blocky but the `run` command does not produce any change on the sense hat.

jwillmer, 2016-09-30:
After a restart the heartbeat example is working on Windows IOT Preview but updating the heartbeat does not work.

gourabSh07, 2016-09-30:
I also have almost same problem Windows IOT image I can't reach the web interface of blockly 
I am using windows build  14393

jwillmer, 2016-09-30:
I was now able to reach the web interface on the stable build after refreshing the build and the app. 

I also managed to get the heartbeat to run after pressing countless times on run but the method is not relayent.

IoTGirl, 2016-10-03:
I don't believe there is a Blockly sample in MS-IoT. Can you please provide a link to the Microsoft Sample you are referring to that is failing?

jwillmer, 2016-10-03:
[IOT Blockly](https://github.com/ms-iot/samples/tree/develop/IoTBlockly)

IoTGirl, 2016-10-03:
Excellent! Thank you for the link. Assigned @rachitb777 to take a look.

rachitb777, 2016-10-03:
@jwillmer Do you see IoTBlocklyBackgroundApp [Startup] listed in the apps page in Windows Device Portal?

jwillmer, 2016-10-03:
Of course, how else would I be able to browse to the website?

alecont, 2016-10-07:
thanks for the feedback! We'll look into this in the next couple of days. We made a couple of changes to the source code here on github which are not reflected yet on the appx deployed by IoT Dashboard.

OwenBrotherwood, 2016-10-09:
@jwillmer 
How about testing with and without Windows IoT Remote Server enabled.
Both the example and the remote server seem to be using the same port, 8000.
That is, do not start the remote server and deploy the sample.

jwillmer, 2016-10-09:
@OwenBrotherwood I can try it but the `Windows IoT Remote Server` is not using port `8000` it uses `8080` as default. Anyhow I think the example should work without any hack.

OwenBrotherwood, 2016-10-10:
@jwillmer : 
Windows IoT Remote Server is NOT started by default: it is up to the administror to start it: a choice that can be affected by the other software chosen to be run on the unit ie NOT a hack
The Web Interface uses 8080 (where one enables Remote Server) 

@alecont The example will work, however the port used should be changed from 8000 to some other port plus a clear indication where in the source code the chosen default can be changed or made as a configurable option.
I have not found out yet where the port 80000 is set: given some information, I can create a pull request.

jwillmer, 2016-10-10:
@OwenBrotherwood I mixed up the `Device Portal` with the `Windows IoT Remote Server`. The `Windows IoT Remote Server` was not active. I deployed a clean install of `Windows IoT` and used the `IoT Dashboard` to deploy `blockly`

OwenBrotherwood, 2016-10-10:
@jwillmer 
Is your status, No Windows IoT Remote Server aktive and Blockly works?

jwillmer, 2016-10-10:
Yes, I see the Blockly web page but I am unable to start any template.

OwenBrotherwood, 2016-10-10:
@jwillmer OK I do not have a sense hat: I want to use the example as a start to integration to https://www.microbit.co.uk/ which has strong links to microsoft
-- and create a pull request for the result if I ever get that far.

"start any template" : do you mean get the template (program) to run and show a result on the sense hat?

jwillmer, 2016-10-10:
@OwenBrotherwood Yes, there is for instance a heartbeat demo that I tryed to run

OwenBrotherwood, 2016-10-10:
@jwillmer @IoTGirl
The line of code that dictates port 8000  (grep -r 8080 *)
I have to double check the port used by Remote server and try and make a pull request with the port as a configurable option with a default that does not conflict with the standard ports used with the startable services, such as Remote Server 
https://github.com/ms-iot/samples/blob/44daa3131eeb71d8b60215115b12f99f403a93d0/IoTBlockly/IoTBlocklyBackgroundApp/StartupTask.cs#L69

@jwillmer 
I will receive a sense in a few days, then I will try to use the sample as intended

OwenBrotherwood, 2016-10-15:
quick update and I may forget to update when finished
A change of port to 8001 did not help.
I have not as yet scanned if the port is already used
I disabled the remote server again as I want to see the Astro work.
I have a blicking red heart

"And it is goodnight from me at the Windows 10 IoT workshop"

jwillmer, 2016-10-16:
@OwenBrotherwood so now you can reproduce the problem?

OwenBrotherwood, 2016-10-16:
@jwillmer The example works if remote server is disabled. 

The device was used for 8 hours at this Maker Faire in Århus, Denmark with the Blocky example 

The last minute of the video shows the stand I was playing at (with 7 inch red screens)

https://www.facebook.com/karinhoegh/posts/10154792942724610

gourabSh07, 2016-10-19:
After disabling  Remote Server i was able to get web interface of iot blockly but heart example is not working for me do i have to do something else to run it ??

OwenBrotherwood, 2016-10-20:
The heart example works for me

jwillmer, 2016-12-01:
Works now with the latest IOT Preview build: `10.0.14931.1000`

closed 2016-12-01 by jwillmer

dnkmk, 2016-12-01:
Shouldn't that be version 14971.1000?

jwillmer, 2016-12-01:
I installed the latest preview build version the Win10 IOT Dashboard found. For me it was 14931 and the admin web panel tells me there are no updates available for the system.

dnkmk, 2016-12-01:
Okay,

Could it be you forgot to choose “Custom” for device type during new device setup?
Not choosing custom results in a production setup I think….