ms-iot/samples #343: IoTConnectClient

PRIMETSS, 2016-10-03:
With IoTHub EastUS (location AUST) with DeviceManagement Preview Enabled
Latest IoTConnectCLient running on RasbberyPi 3

Always get connection error when sending messages to IoTHub

'Status: An error occurred. Please log out and log in with the correct credentials.'

Using hostname from Properties Page and PrimaryKey from Shared Access Policies

PRIMETSS, 2016-10-03:
Worked it out, The Shared Key is for the Device, and you need to create a new Device (Devices Button In Azure Greyed out otherwise)

https://azure.microsoft.com/en-us/documentation/articles/iot-hub-csharp-csharp-getstarted/

IoTGirl, 2016-10-03:
Note that this forum is Windows IoT. Please leverage the resources at [Azure IoT Hub.](https://azure.microsoft.com/en-us/services/iot-hub/)

closed 2016-10-03 by IoTGirl