ms-iot/samples #345: DigitalSignageUAP Video

PRIMETSS, 2016-10-08:
The sample app on a Pi3 is displaying poor video, missing frames.
Is this what other people are seeing?
Is there a plan to get better video performance?

IoTGirl, 2016-10-15:
Hi PRIMETSS,
Have you tried this on other IoT Core platforms?  There are known issues with the GPU on RPi not supporting Direct X and therefore the graphics performance suffers. 
Sincerely,
IoTGirl

PRIMETSS, 2016-10-15:
Not yet, does anyone have recommendations on board on a board they have tried that doesn't have this problem?

IoTGirl, 2016-10-17:
We have a video loop running on an MBM at the office without issue. Basic HD TV resolution, nothing fancy.

IoTGirl, 2016-12-01:
No update for more than 1 month - closing.  Please see the MSDN IoT Forums for more questions regarding IoTCore.

closed 2016-12-01 by IoTGirl