ms-iot/samples #346: Can't deploy OIC Adapter to remote device

dotMorten, 2016-10-12:
Something is wrong with the project settings of the Headless Adapter app in the OIC service. I don't get to pick a remote device.

[image]

The project debugging settings show this:
[image]

Trying to change it to remote device, it just switches back to this unsupported setting when hitting apply.

I'm using Visual Studio 2015 Update 3

dotMorten, 2016-10-12:
Seems like something is borked on my Visual Studio installation. Doesn't work for any C++ background tasks.

closed 2016-10-12 by dotMorten

lflct, 2016-10-12:
I just tried opening that project in VS 2015 Update 3 and I can see the Remote Device option in the list of debuggers, so I believe you're right, the problem is not with the project itself. Make sure you have the latest files, though, just in case.