ms-iot/samples #350: In the invitation to contribute, first link is broken: Aceess Not Authorized

djaus2, 2016-10-19:
You must have a page on the **IoT Core samples page** corresponding to your code <-- This link

IoTGirl, 2016-10-21:
Page is: [https://github.com/ms-iot/samples/blob/develop/README.md](https://github.com/ms-iot/samples/blob/develop/README.md )

IoTGirl, 2016-10-24:
This should be fixed now.

djaus2, 2016-10-24:
OK it works now thx.
Closing

closed 2016-10-24 by djaus2

djaus2, 2016-10-26:
Justine,
I am just trying to get my head around contributing to the ms-iot/Samples repository.
To get things going, I have created a simple project to add, rather than modify an existing project.

OK I have forked and cloned the whole repo and will add my project as a new folder.
I will create a README.md for my project.

My question is what how do I action: You must have a page on the IoT Core samples page<https://developer.microsoft.com/en-us/windows/iot/samples> corresponding to your code.

Thx david

djaus2, 2016-10-26:
So do I need to get an .md file into https://github.com/ms-iot/content/tree/develop/en-US/Samples?
In my case my project is called NoPopUps so I will need https://github.com/ms-iot/content/tree/develop/en-US/Samples/NoPopups.md??
So do I need to for the contents repo, place my NoPopups.md there and get it accepted, its sounding a bit recursive??

From: David Jones [mailto:davidjones@sportronics.com.au]
Sent: Wednesday, 26 October 2016 10:20 PM
To: 'ms-iot/samples' <reply@reply.github.com>
Subject: RE: [ms-iot/samples] In the invitation to contribute, first link is broken: Aceess Not Authorized (#350)

Justine,
I am just trying to get my head around contributing to the ms-iot/Samples repository.
To get things going, I have created a simple project to add, rather than modify an existing project.

OK I have forked and cloned the whole repo and will add my project as a new folder.
I will create a README.md for my project.

My question is what how do I action: You must have a page on the IoT Core samples page<https://developer.microsoft.com/en-us/windows/iot/samples> corresponding to your code.

Thx david

IoTGirl, 2016-10-26:
Hi David,
You need a description for the samples page and a page for your sample, does that clarify?
Sincerely,
IoTGirl