ms-iot/samples #362: Timeout Issue in GpioOneWire

shubhamkumarnayak, 2016-11-10:
When Application starts for the first time I am getting the temperature and humidity and after that I am getting Time Out Issue, again data is coming after 1 minute for a blink and getting the same error
Device: DHT22

jordanrh1, 2016-11-14:
Are you using the 'master' branch or the 'preview' branch? There are known issues with the sample that will be fixed in the next OS version.

shubhamkumarnayak, 2016-11-15:
Master branch

jordanrh1, 2016-11-16:
Please try the 'preview' branch on the latest OS version.

shubhamkumarnayak, 2016-11-17:
ok thanks.

IoTGirl, 2016-11-30:
Hi Shubham,
Do you have an update? 
Sincerely,
IoTGirl

shubhamkumarnayak, 2016-12-02:
yep It is working now, I created a cli from the project and using it from
c# to get the data and it is working perfectly... :)

IoTGirl, 2016-12-05:
Great! I am closing this issue based on your success.

closed 2016-12-05 by IoTGirl