ms-iot/samples #371: What needs to be adjusted to get the SPIDisplay to work with a SH1106 controller?

karmaecrivain94, 2016-11-19:
I've spent several hours (closer to days) trying to get the SPIDisplay sample to work with a 128x64 SH1106 controller, without any success. From what I've found online, the SH1106's RAM is 132x64, even if there are only 128*64 pixels on the screen, and the initialization instructions are slightly different. I've done both changes, but I've still only got the following on the screen:

[image]

Would it be possible to add compatibility for various different screens, including SH1106?

PRIMETSS, 2016-11-19:
Looks at least like your managing to write to Display Ram...
Is the Column Remap ADC set correctly? So Col / Segment is not swapped over?
Looking at Memory Address layout Im calculating its 8 bits (Data) x 64 Line Addresses (LA 00H-3F) x 132 Col (00H->83H) = 67584 bits Per Page x 8 pages  = 540672 bits
So looks like 132 Col x 64 Lines x 8 bit (Display Data)
Im confused about the Common Output mode (COM0-COM63) as seems to suggest relationship between COL & Rows (lines) is swapped??
Are you seeing Page / Column & Line Address right?
Is D0 in the Address Control set to "0" or "1" (which reverses the column order)?
The Specs diagram is a bit confusing with how the Data Bits are written, is it 
When you write data with a Read-Modify-Write command, the column address is automatically incremented, until you finish writing and send the end command.
 
It appears that the ram is divided into 8 Pages, each page is 132 Col x 64 Lines.
Each time you complete a Col write, you need to reset the Page. Col counters

Have you tried just getting 1 page turned on, and see where it shows up on the display, Im assuming if you selected Page 0, and the set Col to 0, and write 8 x 256(FFh) on each write, then and end, you would get a black bar top of display? Each page being 132 Cols of 8 bit (data/lines) over 8 pages which are oriented vertically in bands???

ooeygui, 2016-12-07:
This project might help you out:
https://www.hackster.io/windows-iot/windows-10-iot-core-for-adafruit-spi-touchscreen-bb3795

It isn't exactly for your screen, but give you most of what you need to get going.
Hope this helps!

closed 2016-12-07 by ooeygui