ms-iot/samples #374: "Windows IoT Core Project Templates for VS 2017" tree position is wrong.

Tak1wa, 2016-11-28:
At the Project Creation Wizard
It should be [Visual C #] -> [Windows Universal], but it is under [Visual C#] -> [Windows Classic Desktop]

bfjelds, 2016-11-30:
Thank you for the feedback.  Are you building the templates yourself?  What version of VS2017 are you installing the templates for (VS2017 Preview 4, Preview 5, RC)?

Our goal will be for it to be a peer to Windows Universal.  That said, we haven't finished working on compatibility with VS2017 yet.

bfjelds, 2016-12-02:
It looks like the RC has two states ... 

1. If the C++ UWP templates are installed, they will show up under **Templates > Visual C++ > Windows > Windows Universal**.

1. If the C++ UWP templates are NOT installed, a node for installing the templates will show up as **Templates > Visual C++ > Windows Universal**.

IoTGirl, 2016-12-05:
Hi Tak1wa,
Based on Brian's check in I am closing this issue.  Please reopen if you feel the problem has not been addressed.
Sincerely, IoTGirl

closed 2016-12-05 by IoTGirl