ms-iot/samples #376: Blockly sample web server uses same port as the Remote Server

juunas11, 2016-12-01:
I was trying to get the IoTBlockly sample application installed from IoT Core Dashboard onto my Raspberry Pi 3 and had an issue with the background service crashing. I cloned the repo and ran it from VS. The result was an error like this:

    Exception thrown: 'System.Runtime.InteropServices.COMException' in mscorlib.ni.dll
    WinRT information: Only one usage of each socket address (protocol/network address/port) is normally permitted.

I got a remote PowerShell open and ran netstat to find out what is taking the port. Turns out it was `NanoRDPServer.exe`, which is the "Remote Server" you can enable from Windows Device Portal. After disabling the Remote Server, everything worked fine.

*Could we change the port that IoTBlockly uses?* And also probably audit the other projects that listen for TCP connections in case something else is also using port 8000 or another port used by a basic service on the device.

ooeygui, 2016-12-07:
Oh... Good catch.

@alecont ?

IoTGirl, 2016-12-07:
Hi juunas11,

The IOT Core team has accepted this request into their backlog (10105126) but as this team works on the agile model they can not provide a timeline for it to be reviewed/addressed.

Sincerely,
IoTGirl

closed 2016-12-07 by IoTGirl