ms-iot/samples #389: SpeechSynthesizer Voice How to install other languages

Yuexs, 2016-12-25:
SpeechSynthesizer Voice How to install other languages

How to install other language voice synthesis engine, I tried to extract from windows10 to IOT, but still can not be synthesized.

I now develop smart home voice synthesis and recognition dependence is high, I do not know how to solve this problem, very grateful to such a good system

Yuexs, 2016-12-25:
Iot OS version:10.0.14393.576

Yuexs, 2016-12-25:
I tried to extract the windows10 voice to the device

And try to set up the registry

HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Speech_OneCore\Voices\Tokens

Still no effect

Ask for help

IoTGirl, 2016-12-26:
Closing duplicate - See [https://social.msdn.microsoft.com/Forums/en-US/d09494a0-f0af-434a-9e77-72da7500a8bb/windows-10-iot-speech-languages?forum=WindowsIoT](https://social.msdn.microsoft.com/Forums/en-US/d09494a0-f0af-434a-9e77-72da7500a8bb/windows-10-iot-speech-languages?forum=WindowsIoT) and other similar threads

closed 2016-12-26 by IoTGirl