ms-iot/samples #410: Error while deploying appx file.

kikinewton, 2017-02-14:
I tried to install the appx file of HelloBlinky on a raspberry pi 3 and and am getting this error. Any help will be appreciated. 
Failure reason: Windows cannot install package Blinky_1.0.1.0_arm__1w720vyc4ccym because this package depends on a framework that could not be found. Provide the framework "Microsoft.NET.CoreRuntime.1.0" published by "CN=Microsoft Corporation, O=Microsoft Corporation, L=Redmond, S=Washington, C=US", with neutral or ARM processor architecture and minimum version 1.0.23819.0, along with this package to install. The frameworks with name "Microsoft.NET.CoreRuntime.1.0" currently installed are: {}. Failure text: Package failed updates, dependency or conflict validation. (0x80073cf3)

bfjelds, 2017-02-14:
How are you trying to deploy the APPX file?  Are you including the dependency APPX files?

As an example, from the Device Portal, to deploy HelloBlinky, you should build the APPX and then enter:

1. Enter App Package = HelloBlinky\CS\AppPackages\Blinky_1.0.0.0_Debug_Test\Blinky_1.0.0.0_arm_Debug.appx
1. Enter Certificate = HelloBlinky\CS\AppPackages\Blinky_1.0.0.0_Debug_Test\Blinky_1.0.0.0_arm_Debug.cer
1. Click Add Dependency 2x
    + HelloBlinky\CS\AppPackages\Blinky_1.0.0.0_Debug_Test\Dependencies\ARM\Microsoft.NET.CoreRuntime.1.0.appx
    + HelloBlinky\CS\AppPackages\Blinky_1.0.0.0_Debug_Test\Dependencies\ARM\Microsoft.VCLibs.ARM.Debug.14.00.appx
1. Click Go button

kikinewton, 2017-02-16:
Thanks for the help. Your solution worked.

  2.  Enter Certificate = HelloBlinky\CS\AppPackages\Blinky_1.0.0.0_Debug_Test\Blinky_1.0.0.0_arm_Debug.cer
  3.  Click Add Dependency 2x
  4.  Click Go button

IoTGirl, 2017-02-17:
Closing based on our confirmation

closed 2017-02-17 by IoTGirl