ms-iot/samples #412: Is there a guide available for this project?

florindpreda, 2017-02-16:
Hello, 

I'm new to the Microsoft IoT Core. I managed to setup and run the Blinky app but now I'm looking on how to connect my Raspberry PI 3 to Azure and PowerBI.

I found this project which I would like to run but I'm not sure what are the dependencies for it. Is there a guide page or file available?

Thanks

IoTGirl, 2017-02-17:
There is a blog at [https://blogs.msdn.microsoft.com/iot/2016/01/26/using-power-bi-to-visualize-sensor-data-from-windows-10-iot-core/](https://blogs.msdn.microsoft.com/iot/2016/01/26/using-power-bi-to-visualize-sensor-data-from-windows-10-iot-core/)

IoTGirl, 2017-03-13:
@florindpreda  As you have no further questions I am closing your issue.

closed 2017-03-13 by IoTGirl