ms-iot/samples #418: AppServiceSharedNotepad/C#: Client closing issue

LeonardoDaga, 2017-02-27:
Hello, 
I found that when one of the clients (both the background or the foreground) stops, the server is killed (closed) even if another client is attached.
Is this a desired behavior? Is it possible to check if another client is connected and eventually avoid to close the NotepadServer?

IoTGirl, 2017-02-27:
Hi Leonardo,
As you have the sample are you able to step through the code and identify the changes you would like to make for your implementation?  These are samples and not robust solutions so likely they would need more work before being ready for a real world product.
Sincerely,
IoTGirl

LeonardoDaga, 2017-02-28:
Hello IoTGirl,
I really don't know where to change the code to change the behavior. I already described the situation also here: 
https://social.msdn.microsoft.com/Forums/en-US/59e959dc-52eb-4034-89fb-bee7812fc010/appservicesharednotepadc-example-survival-of-the-server-after-a-client-closes?forum=WindowsIoT

Sorry for posting twice :-(

IoTGirl, 2017-03-01:
Hi Leonardo,
This is the correct forum for that posting as it relates to a sample within this repository. 
Sincerely,
IoTGirl

paulmon, 2017-03-02:
See this pull request.  https://github.com/ms-iot/samples/pull/421 
It's important to register the Canceled event for each AppService instance and Complete the deferral if a deferral is still held.

closed 2017-03-02 by bfjelds