ms-iot/samples #419: SerialUART: ReadAsync didn't return any data if some UI-related code disabled

Leeway213, 2017-03-01:
I tried to modify the connect button click listener like this:
```
private async void comPortInput_Click(object sender, RoutedEventArgs e)
        {
            var selection = ConnectDevices.SelectedItems;

            if (selection.Count <= 0)
            {
                status.Text = "Select a device and connect";
                return;
            }

            DeviceInformation entry = (DeviceInformation)selection[0];         

            try
            {                
                serialPort = await SerialDevice.FromIdAsync(entry.Id);

                // Disable the 'Connect' button 
                comPortInput.IsEnabled = false;

                // Configure serial settings
                serialPort.WriteTimeout = TimeSpan.FromMilliseconds(1000);
                serialPort.ReadTimeout = TimeSpan.FromMilliseconds(1000);                
                serialPort.BaudRate = 115200;
                serialPort.Parity = SerialParity.None;
                serialPort.StopBits = SerialStopBitCount.One;
                serialPort.DataBits = 8;
                serialPort.Handshake = SerialHandshake.None;

                // Display configured settings
                status.Text = "Serial port configured successfully: ";
                //status.Text += serialPort.BaudRate + "-";
                //status.Text += serialPort.DataBits + "-";
                //status.Text += serialPort.Parity.ToString() + "-";
                //status.Text += serialPort.StopBits;

                // Set the RcvdText field to invoke the TextChanged callback
                // The callback launches an async Read task to wait for data
                rcvdText.Text = "Waiting for data...";

                // Create cancellation token object to close I/O operations when closing the device
                ReadCancellationTokenSource = new CancellationTokenSource();

                // Enable 'WRITE' button to allow sending data
                sendTextButton.IsEnabled = true;

                Listen();
            }
            catch (Exception ex)
            {
                status.Text = ex.Message;
                comPortInput.IsEnabled = true;
                sendTextButton.IsEnabled = false;
            }
        }
```

As above, if disable the code:
```
                //status.Text += serialPort.BaudRate + "-";
                //status.Text += serialPort.DataBits + "-";
                //status.Text += serialPort.Parity.ToString() + "-";
                //status.Text += serialPort.StopBits;
``` 
ReadAsync couldn't return any data from serial device. 
Why?

jordanrh1, 2017-03-04:
It doesn't make any sense. These lines should have no effect on the serial port operation. Could there be another reason ReadAsync isn't returning data? How is data being written to the port?

Leeway213, 2017-03-13:
Recent update: This issue existed when I was using a z-tek USB to serial cable to debug but disapeared when using another one which is SoC internal UART. Maybe it's a hardware bug of USB to serial device?