ms-iot/samples #423: Why I cannot download the sample code

CaseyLu, 2017-03-14:
Anybody has such problems?

CaseyLu, 2017-03-14:
PS: My win 10 IoT is on Raspberry pi 3

IoTGirl, 2017-03-16:
Hi CaseyLu,
There is an existing issue at [https://social.msdn.microsoft.com/Forums/Windowsserver/en-US/57d40f2a-399a-46d9-91da-6e6c273baf13/cbs-status-failure-unable-to-debug-the-deploy-driver-to-rpi3?forum=WindowsIoT](https://social.msdn.microsoft.com/Forums/Windowsserver/en-US/57d40f2a-399a-46d9-91da-6e6c273baf13/cbs-status-failure-unable-to-debug-the-deploy-driver-to-rpi3?forum=WindowsIoT ) that indicates this error may have been the result of mis-matched tools.  Can you confirm the actual version number of both the device OS and ADK/SDK/WDK you have installed?
Sincerely,
IoTGirl

IoTGirl, 2017-03-23:
Hi @CaseyLu,
As you have not responded I would ask that you follow up on the forums thread above if you have more questions regarding this issue.
Sincerely,
IoTGirl

closed 2017-03-23 by IoTGirl