ms-iot/samples #432: could not find assembly 'platform.winmd'

cyberh0me, 2017-03-22:
hello
what went wrong if building fails with C1107 

> could not find assembly 'platform.winmd': please specify the assembly search path using /AI or by setting the LIBPATH environment variable

this occurs with vs2017 and also vs2015('Windows.winmd' instead of 'platform.winmd')
br
Andre

IoTGirl, 2017-03-23:
Hi Andre,

Have you tried repairing the toolsets in question? ADK/SD?WDK?  Also can you confirm that the versions all match?

Sincerely,
IoTGirl

cyberh0me, 2017-03-23:
hello
ADK and WDK should not be needed and SDK from both machines is the one included in the VS Editions and both are new installations
br
Andre

IoTGirl, 2017-04-04:
Hi Andre,
Is this still an open issue for you?
Sincerely,
IoTGirl

cyberh0me, 2017-04-04:
even if i don't have an answer or solution its now on the big list of not working iot projects
br
Andre

closed 2017-04-04 by cyberh0me