ms-iot/samples #437: USB-Serial Freezes OS when buffer is full (Blue Screen)

MyersMade, 2017-03-23:
Using Hardware or Software Handshaking on an external USB-Serial (FTDI) device:

When the receiving devices holds it's RTS low, the StoreAsync() should time out based on the .WriteTimeout parameter in the SerialDevice, return a 0 and allow me to react appropriately.

In IOT Core however, the StoreAsync() never times out, and will even lock up the entire OS and cause a blue screen watchdog violation after a while.

I believe there is a serious issue with the Microsoft-provided driver for USB-Serial devices.

I've detailed the issue further and supplied sample code here:

[https://github.com/MyersMade/PiSerialCrashTest](https://github.com/MyersMade/PiSerialCrashTest)

IoTGirl, 2017-03-23:
Hi @MyersMade ,

I will ask someone to review this issue but please be aware that there are a number of issues with the USB hardware Stack in the RPi.  Have you seen this issue on any other platform?

Sincerely,
IoTGirl

MyersMade, 2017-03-23:
No, Pi and Windows Desktop is all I have available.

MyersMade, 2017-03-30:
Jordan, have you had a chance to look at this?

IoTGirl, 2017-04-04:
This issue has been cross posted - [https://social.msdn.microsoft.com/Forums/en-US/a299cdbf-6087-4f37-baa9-43e9f8cea2af/serial-rs232-datawriterstoreasync-causes-blue-screen-in-windows-iot-core?forum=WindowsIoT#6b167353-a856-4ab2-b372-31eea9d12cc6](https://social.msdn.microsoft.com/Forums/en-US/a299cdbf-6087-4f37-baa9-43e9f8cea2af/serial-rs232-datawriterstoreasync-causes-blue-screen-in-windows-iot-core?forum=WindowsIoT#6b167353-a856-4ab2-b372-31eea9d12cc6)

embedded101, 2017-04-04:
@MyersMade,

There are known issues with certain FTDI USB-to-RS232 chip (I don't remember the details.).
A while back, I was using a batch of USB-to-RS232 adapter from FTDI for a community project using Windows 10 as the dev machine.  I got report from some participants the FTDI adapters had problems.
I reached out to my contact at FTDI at the time and was told there was known issue with the USB-to-RS232 adapters we were using. 

Sorry, I don't have the affected FTDI) part number.  You should check to make sure you are not using a problematic device.  

If the USB-to-Serial device is external to the controller, I would try a different chip to see whether the same problem occur.

MyersMade, 2017-04-04:
Thanks for your response.  My hardware is an external USB-RS232 adapter

I've tried two different models:

GEARMO FTDI-LED
Chipset FTDI FT232RL
(tried 2 different units)

Unknown Mfg
Chipset FTDI FT232RL

Both devices I got the same results.

[image]
[image]
[image]
[image]

[image]
[image]

I guess I had assumed the hardware was okay, since they work fine in Windows Desktop.

My gut tells me that the issue is coming from whatever driver that IOT ships with for the FTDI chipsets.

Does anything stand out from what I've posted here?