ms-iot/samples #445: The type or namespace name 'ApplicationInsights' does not exist in the namespace 'Microsoft' (are you missing an assembly reference?)

baotrung1309, 2017-03-29:
Hi all, 
I'm trying to run the sample "ZigbeeAdapter" in Alljoyn folder and I also had the same problem. The code is as following:
public App()
{
Microsoft.ApplicationInsights.WindowsAppInitializer.InitializeAsync(
Microsoft.ApplicationInsights.WindowsCollectors.Metadata |
Microsoft.ApplicationInsights.WindowsCollectors.Session);
this.InitializeComponent();
this.Suspending += OnSuspending;
}
The error log: The type or namespace name 'ApplicationInsights' does not exist in the namespace 'Microsoft' (are you missing an assembly reference?)

Did anyone solve this problem?? Thanks in advanced!

best Regards,
Trung.

GouthamiJJ, 2017-03-29:
Hi all,

Even I'm also facing the same issue please give a solution to solve it.

Regards,
Gouthami

summoncse, 2017-03-29:
I have installed Microsoft.ApplicationInsights.WindowsApps dependency but I am not able to discover any ZigBee Device.

zhuridartem, 2017-04-01:
The missing ApplicationInsights should be resolved with the pull request #455. 
To work with ZigBee devices, please see our tutorial at https://developer.microsoft.com/en-us/windows/iot/Samples/ZigBeeAdapter. You will need to setup a ZigBee network first before bridging it to AllJoyn.

closed 2017-04-03 by zhuridartem