ms-iot/samples #474: build issue with I2cTestTool/

rajnishc8, 2017-04-14:
Hello,
 I get following error when I try to build I2CTestTool. 

1>------ Build started: Project: I2cTestTool, Configuration: Release x64 ------
1>main.cpp
1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\include\agile.h(16): fatal error C1083: Cannot open include file: 'wrl\client.h': No such file or directory
1>Done building project "I2cTestTool.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

Thanks,
--Rajnish

IoTGirl, 2017-04-17:
Hi @rajnishc8,

The message 'wrl\client.h' says you are missing the "wrl" or Windows Runtime Library files.  Are you sure you have installed all the required tools to use the IoT Core samples?  

Sincerely,
IoTGirl

rajnishc8, 2017-04-18:
Hi @IoTGirl,
I have vs2015, vs2017, EWDK installed.

To get it to build successfully in VS2017, I had to set
Additional Include Directories to:
C:\Program Files (x86)\Windows Kits\10\Include\10.0.14393.0\um;C:\Program Files (x86)\Windows Kits\10\Include\10.0.14393.0\shared;C:\EWDK\Program Files\Windows Kits\10\Include\10.0.14393.0\winrt
and
Additional Dependencies to:
C:\Program Files (x86)\Windows Kits\10\Lib\10.0.14393.0\um\x64\onecoreuap.lib;$(AdditionalDependencies)

Thanks,
--Rajnish

IoTGirl, 2017-04-21:
Thanks for the update Rajnish!

paulmon, 2017-04-21:
If you have the current code in github it shouldn't be using the 14393 version of the SDK, it should be using 15063.  It should also be using the v141 compiler settings rather v140.

rajnishc8, 2017-04-22:
Please refer to screen shot given below
SDK version is :  10.0.15063.0 .
[image]

It can be seen in the screen capture that it does not build with vs2017 and sdk 10.0.15063.0 . 

Thanks,
--Rajnish

IoTGirl, 2017-04-25:
Hi @rajnishc8,
Do you have VS 2015 also installed on this machine?  The packages seem to only attribute to one or the other not both versions of VS so if you have VS2015 installed, it may be that the packages are actually attached to that prior version of VS rather than VS 2017.
Sincerely,
Justine