ms-iot/samples #483: OpenCV example Visual Studio 2017

akeilox, 2017-05-13:
I am trying to compile the OpenCV example on Visual Studio 2017 with c++ dev installed.

During build it complains cant locate the opencv_core300d,lib which I presume needs to be compiled from ms opencv fork at **https://github.com/Microsoft/opencv**

but I cant get that fork to compile under Visual Studio 2017 either, enquired the same under **https://github.com/Microsoft/opencv/issues/78**

Is there any fellow VS 2017 user who manage to get the OpenCV to compile/work on the windows IOT?

akeilox, 2017-05-13:
following up on the post above, i might have a **hacky** solution;

after installing c++ dev tools for vs 2015, the project openCV still did not  compile.
So I have tried renaming all the references to 140 to 141 in the .vcxproj files

After which I have changed the SDK version of the openCV fork to 10.0.15 (latest)

then i have compiled **both** debug and release versions for ARM and x86, to get the libs (w/o ..d) build.

After all of the above 'hacks' i was able to compile the openCV for arm and x86 within VS 2017.

Finally I have build the OpenCV iot example, but noticed it complains about App started but activation request failed -  when i try to release it to Raspberry Pi 3 running latest IOT - assuming there is a trick here im missing?
Hopefully someone can clarify.

Exact error when debugging on remote machine, after transferring the app;

Unable to activate Windows Store app
'4277356b..........'
The OpenCVExample.exe process started but the activation request failed with error 'The app didnt start'

Would appreciate if someone can shed light on this.

IoTGirl, 2017-05-16:
Hi @akeilox ,
Please research this with the repository owners at [https://github.com/Microsoft/opencv/issues](https://github.com/Microsoft/opencv/issues) 
Sincerely,
IoTGirl

closed 2017-05-16 by IoTGirl

JimSEOW, 2017-07-25:
@IoTGirl  We need to have opencv_core300d.lib and opencv_core300,lib available through Nuget. So others do not have to repeat the the process of how to get it works