ms-iot/samples #489: IoTOnboarding doesn't build

btusca, 2017-05-29:
Hi!
I'm trying to make it work the CompanionApp samples but the SoftAP's ssid doesn't appears in my wireless list. I followed the instructions about adding a registry. Then, I read an issue where explains that I have to redeploy a new IoTOnboarding app to enable SoftAP.

When I tried it, VS2017 show me this:
[image]

How should I proceed?

The OS Build is Windows Insider Preview 16193 and the device is a Raspberry Pi 3.

Thanks!

IoTGirl, 2017-05-31:
Hi @btusca,
The SoftAP should be enabled by default. Are you attempting to disable it?
Sincerely,
IoTGirl

btusca, 2017-07-09:
yes, it is enabled,
I just want to change the wifi configuration without using an ethernet cable or the device portal.