ms-iot/samples #501: Setting-Up sensortag...

phardesty, 2017-07-07:
ENVIRONMENT:
**What Computers, OS and devices are involved with this repro? How are they connected?**
We are using the sample below with the Texas Instruments I SimpleLink SensorTag Development Kit Featuring the CC2650 wireless MCU. When running on a Windows 10 creators update and Windows phone Lumia950 XL running Windows 10 1703 build number 10.0.15228.0
REPRO STEPS:
What steps would someone else need to follow to see the issue?
1.	Compile the sample app on Dell laptop, Using VS 2017 community edition.
2.	Run locally or on Windows phone device.
3.	Follow steps in the app to pair the GATT device.
4.	App will acknowledge and state setting-up device.
5.	Device will display active connection (pairing light turn-off).
6.	Wait for several minutes, nothing in the app changes at all.
7.	Optionally used disconnect button in the app
a.	Device enters pairing mode again lights start flashing

RESULTS:
**What did you Expect to happen?**
We expected the sensor to return values to our device it returned nothing. It just displayed setting up sensor tag.
**What did Actually happen?**
Both devices hang at setting up device after going through the pairing process in the sample application
**NOTES:**
Any other details/workarounds potentially related info you can provide?
Link to source code https://github.com/ms-iot/samples/tree/develop/BluetoothGATT/CS

IoTGirl, 2017-07-10:
Hi @phardesty,

Have you looked at David Jones's post regarding the Ti Sensor tags?  Maybe it would help.
[http://www.embedded101.com/Blogs/David-Jones/entryid/782/Bluetooth-BLE-Creators-Edition-and-RPi3-issue-](http://www.embedded101.com/Blogs/David-Jones/entryid/782/Bluetooth-BLE-Creators-Edition-and-RPi3-issue-)

Sincerely,
IoTGirl

wirelessdreamer, 2017-07-22:
This example is not working for me either.
Windows Build: 14393.1480
Bluetooth Adapter ASUS USB-BT400
Visual Studio 2017
SensorTag with CC2650 MCU

On startup the sensor tag shows up in the device list.
it takes about 10 seconds for pair and connect to change color, so it can be clicked on.
after clicking on it it says Setting up Sensor tag. then throws an exception on line 444 of MainPage.xasl.cs :                    GattReadResult result = await characteristicList[0].ReadValueAsync(BluetoothCacheMode.Uncached);

if I comment out 441-447 it starts up, and temperature, humidity, and keypress are enabled on startup. when I select Accel, Magnometer, or Gyro the enable button next to them never enables, and I am unable to read data from them.

I am missing something, or is the example currently broken?